Xinyuan-LilyGO/T-SIM7600X #119: No GPS being received

Kyliroco, 2025-01-29:
Hello I'm new with this card but I can't connect to the gps I used the code to test all the functionality by changing the apn and the gnss_mode and everything works except the gps any idea where this can come from? I obviously connected the antenna.
I've looked at other gps issues but haven't found anything conclusive.
Is there something to set that I've forgotten?

[debug .txt](https://github.com/user-attachments/files/18592386/debug.txt)

PsySc0rpi0n, 2025-02-03:
What about the Serial port? What does is shows? Anything useful?
Are you inside walls or in a place the satellites cannot be locked or something?

Kyliroco, 2025-02-03:
Thanks for the answer, after searching for a while the problem is just that you have to put the gps outside any building, even near a window. It doesn't work for the launch, but once it's launched you can go back into the buildings and it works.

closed 2025-02-03 by Kyliroco

PsySc0rpi0n, 2025-02-03:
> Thanks for the answer, after searching for a while the problem is just that you have to put the gps outside any building, even near a window. It doesn't work for the launch, but once it's launched you can go back into the buildings and it works.

Great that you figured it out.
However, in my short experience, I have a LilyGo TSIM7600X and if I place the device next to the window, I usually can get lock on satellites. If I move it, even 1m inside the room, it will eventually lose the lock and no GPS data is retrieved. But if that method works for you, nice. :)