Xinyuan-LilyGO/T-SIM7600X #122: Firmware Upgrade Failed

abdullahbaa5, 2025-03-02:
Facing CP Download Failed! Parition diff!

[image]

When updating firmware of SIMCOM-7600G-H Module.
I have two modules, one was updated just fine but the other one is failing.

The documentation for the simcom update app says the following:

[image]

Could .qmbn file be uploaded along with the firmware as well on the update guide page?

lewisxhe, 2025-03-03:
Same modem? Use at+simcomati to check,

and make sure the PN number on the modem label must be the same

abdullahbaa5, 2025-03-03:
I have:

SIMCOM 7600G-H R.2 Module:
P/N: S2-1097D-Z31E3
Faced error with this one

But for
SIMCOM 7600G R.2 Module:
P/N: S2-1097C-Z31FP
It updated just fine

lewisxhe, 2025-03-03:
- [SIMCOM-7600G-H PN:S2-1097D-Z31E3](https://drive.google.com/drive/folders/11YthuETFAIiLzhQ2mxGtca27JOckc2oP?usp=sharing)
- [SIMCOM-7600G-H PN:S2-1097C-Z31FP](https://drive.google.com/drive/folders/11pfGCjkZFXo1AnTbNawqvSGudWjKn_26?usp=sharing)

abdullahbaa5, 2025-03-03:
> * [SIMCOM-7600G-H PN:S2-1097D-Z31E3](https://drive.google.com/drive/folders/11YthuETFAIiLzhQ2mxGtca27JOckc2oP?usp=sharing)
> * [SIMCOM-7600G-H PN:S2-1097C-Z31FP](https://drive.google.com/drive/folders/11pfGCjkZFXo1AnTbNawqvSGudWjKn_26?usp=sharing)

Thank you!

Does these firmware have the fix where if you are writing data to the audio driver and the call is ended while data is writing, the next call you will not be able to write data (will get timed out) until and unless you replug the usb device.

lewisxhe, 2025-03-03:
I don't know about this, you can test it first. This is the latest firmware

abdullahbaa5, 2025-03-03:
> I don't know about this, you can test it first. This is the latest firmware

the [SIMCOM-7600G-H PN:S2-1097D-Z31E3](https://drive.google.com/drive/folders/11YthuETFAIiLzhQ2mxGtca27JOckc2oP?usp=sharing) seems to have fixed it for the -H module, however i noticed the issue, it is that there is still data in the buffer of the device that waits till next call to be played.

Is there an AT command to clear the audio buffer on-device?

*Note: I am performing AT+CPCMREG=0 when there is no call, and AT+CPCMREG=1 when call starts

lewisxhe, 2025-03-04:
You need to use a recorder to record the problematic areas so that simcom can analyze the problems.
There is a README file inside, please read the steps in it
[DiagGrab](https://drive.google.com/file/d/11x_J4RiHb168PCx9OwyLoqq1MEj0l18P/view?usp=sharing)