Xinyuan-LilyGO/T-SIM7600X #123: A fatal error occurred: Failed to connect to ESP32: No serial data received.

norielcahilig, 2025-03-24:
Can you help us out, we can't upload our new codes becuase this serial data error. Thank you so much

lewisxhe, 2025-03-24:
Lower the baud rate to 115200,
connect GPIO0 to GND, and then power on. You should be able to write now.

norielcahilig, 2025-03-24:
is there a way we could factory reset the board, seems like we're lost in
track huhu

lewisxhe, 2025-03-26:
Can you upload a sketch now?

The factory program is the ATDebug sketch

norielcahilig, 2025-03-28:
yes thank u so much for your help we could upload a sketch again

do u have a partial or full sketch on how we could connect a esp32 as
remote to lilygo t-sim7600G through bluetooth? any help would be always
appreciated

lewisxhe, 2025-04-01:
Bluetooth connection? Do you need a Bluetooth example?

The Bluetooth example is already included in the examples that come with esp32

norielcahilig, 2025-04-01:
good day sir, are u referring to the repositories? upon checking, we can't
find any BT sketch there. thank you for responding :))

lewisxhe, 2025-04-01:
[image]

norielcahilig, 2025-04-01:
oh in arduino, i see thank you so much

norielcahilig, 2025-04-04:
Good day, it's us again, we are actually making progress in our project
with your help.

 Do you know what is the possible issue why SMS don't send a message with a
link?
Example: The message we are trying to send MUST have a google maps link but
it seems that it doesn't send when it contains a link.

Thank you Sir !!

lewisxhe, 2025-04-07:
How did you send it? Give a short example

norielcahilig, 2025-04-07:
[image: image.png]

norielcahilig, 2025-05-07:
hello sir we are still working with the project, this time the sim7600G
slows down in enabling gps, what should we do? we really need ur help

lewisxhe, 2025-05-08:
Please describe the problem in detail and how to reproduce it

norielcahilig, 2025-05-08:
so connected another microcontoller to the sim7600G thru bluetooth that
will serve as our remote, after the bluetooth initiated, connected, the
modem also initiated, then we're stocked in "enabling gps/gnss/glonas", and
it will never proceed

lewisxhe, 2025-05-08:
Have you tested this example? It doesn't matter whether the modem and Bluetooth are enabled, they are independent
https://github.com/Xinyuan-LilyGO/T-SIM7600X/blob/master/examples/GPSDebug/GPSDebug.ino

norielcahilig, 2025-05-08:
yea, it'll only work once but next trials, the process will be either
unhandled, or it will be stocked in enabling process

lewisxhe, 2025-05-08:
Please give me a test step, the situation you said should not occur

lewisxhe, 2025-05-08:
The picture you sent cannot be viewed, it is text

norielcahilig, 2025-05-08:
https://drive.google.com/file/d/1IKw9YCaKgZGSuwBuR3q02taunpQ90Sgc/view?usp=sharing

here sir

lewisxhe, 2025-05-08:
Enable DUMP_AT_COMMANDS to get more information

norielcahilig, 2025-05-08:
how sir? sorry

lewisxhe, 2025-05-08:
Enable DUMP_AT_COMMANDS so you can get more information

https://github.com/Xinyuan-LilyGO/T-SIM7600X/blob/f6b8dfe009b664c279c63f1042acca4ebdb3ef70/examples/AllFunctions/AllFunctions.ino#L22

norielcahilig, 2025-05-08:
what are we expecting to see sir? after we enable dump at commands

lewisxhe, 2025-05-08:
You can see the commands sent to the modem and the results returned by the modem, and whether it is in the positioning state.

norielcahilig, 2025-05-08:
thank you sir

AT+CGPS=1
AT+CGPS=1
ERROR

these seems to be the error? what does this mean?

lewisxhe, 2025-05-08:
AT+CGPS? Check whether it is turned on?

lewisxhe, 2025-05-08:
Before confirming that the GPS is working properly, use ATDebug to perform AT interaction to test the GPS.

norielcahilig, 2025-05-08:
https://drive.google.com/file/d/1pO8pnjeObrYoSOdqJcPkEXiGkBTFtThL/view?usp=sharing

ran ATDebug and send AT Commands in serial monitor, here's what happen sir,
seems like it is still an error, but other AT was OK

lewisxhe, 2025-05-08:
```
» AT+CGPS?
« AT+CGPS?

+CGPS: 0,1

OK
» AT+CGPS=1
« AT+CGPS=1

OK
» AT+CGPS?
« AT+CGPS?

+CGPS: 1,1

OK
» AT+CGPSINFO
« AT+CGPSINFO

+CGPSINFO: ,,,,,,,,

OK
» AT+CGPSINFO
« AT+CGPSINFO

+CGPSINFO: ,,,,,,,,

OK
» AT+CGPSINFO
« AT+CGPSINFO

+CGPSINFO: ,,,,,,,,

OK
» AT+CGPSINFO
« AT+CGPSINFO

+CGPSINFO: XXXX.277539,N,XXXX.601710,E,080525,073520.0,62.3,0.0,

OK
» AT+CGPS?
« AT+CGPS?

+CGPS: 1,1

OK

```

Are you serious? The command you sent is invalid. Have you read the AT command manual?

norielcahilig, 2025-05-08:
apologies, we'll try these commands sir

norielcahilig, 2025-05-08:
15:44:27.101 -> AT+CGPS?

15:44:27.147 -> +CGPS: 1,1

15:44:27.147 ->

15:44:27.147 -> OK

15:44:43.426 -> AT+CGPSINFO

15:44:43.458 -> +CGPSINFO:
1436.753711,N,12105.122952,E,080525,074442.0,31.5,0.0,

15:44:43.533 ->

15:44:43.533 -> OK

15:44:59.248 -> AT+CGPS?

15:44:59.248 -> +CGPS: 1,1

15:44:59.248 ->

15:44:59.295 -> OK

here's what we've got sir, sorry for the inconvenience

norielcahilig, 2025-05-08:
15:46:09.393 ->

15:46:09.393 -> OK

15:51:38.146 -> AT+CGPS=1

15:51:38.146 -> ERROR

BUT ON AT+CGPS=1, IT IS ERROR

norielcahilig, 2025-05-08:
please kindly help us sir :((

norielcahilig, 2025-05-08:
> If GPS is already on, sending AT+CGPS=1 will definitely return an error.
>
> Send AT+CGPS? to check whether it is on. If it is on, you do not need to
> send AT+CGPS=1.

15:44:27.101 -> AT+CGPS?

15:44:27.147 -> +CGPS: 1,1

15:44:27.147 ->

15:44:27.147 -> OK

this is when i send AT+CGPS

lewisxhe, 2025-05-08:
Once it is turned on, there is no need to send a command to turn it on again, do you understand?

norielcahilig, 2025-05-08:
Can we upload our program again? Is the issue already fixed after that?
thank you so much, we're very sorry for the convenience

lewisxhe, 2025-05-08:
You need to understand what needs to be done to use sim7600, then you can upload your own program

lewisxhe, 2025-05-08:
I can't answer your question, I don't know what you did.

norielcahilig, 2025-05-08:
thank you for entertaining our questions, sorry again for the inconvenience