Xinyuan-LilyGO/T-SIM7600X #127: Loosing Data Connection from MQTT post Sending Credit Balance SMS text

droidblastnz, 2025-04-30:
@lewisxhe is there a process for being connected to MQTT then sending a SMS Credit Balance Check? This seems to be making the modem become out of control until a hard reboot. 

I noted [same issue](https://github.com/vshymanskyy/TinyGSM/issues/827)

lewisxhe, 2025-05-05:
Which method do you use to connect to MQTT?

droidblastnz, 2025-05-05:
Using the tinygsm library 

`boolean mqttConnect()                                                              
{
  SerialMon.print("Connecting to ");
  SerialMon.print(broker);
  SerialMon.println("");                                                           
  boolean status = mqtt.connect("test01", "user01", "password123");

  if (status == false) {
    SerialMon.println("MQTT Connect fail");
    return false;
  }
  SerialMon.println("MQTT Connect success");
 
  //If GRPS Connected
  if ((status == true) && modem.isGprsConnected()) { 
  }

 // MQTT is not connected check
if (!mqtt.connected()) {
    SerialMon.println("MQTT NOT CONNECTED! ");
    SerialMon.print("Disconnecting from: ");
    SerialMon.println(broker);
    mqtt.disconnect();                                                             
    delay(500);
    SerialMon.print("Network state is: ");
    SerialMon.println(modem.isGprsConnected());

// Reconnect every 10 seconds
    uint32_t t = millis();
    if (t - lastReconnectAttempt > 10000L) {
        lastReconnectAttempt = t;
        if (!modem.isGprsConnected()) {                                            
            SerialMon.println("GPRS not connected, reconnecting...");
            if (!modem.gprsConnect(apn, gprsUser, gprsPass)) {
                SerialMon.println("GPRS reconnect failed");
                delay(10000);
                return;
            }
        }
        if (mqttConnect()) {
            lastReconnectAttempt = 0;
        }
    }
    delay(100);
    return;
}
    mqtt.loop();`

I use this function to get the credit balance

`// Check Balance send to MQTT every ...
void Check_Balance() {
    CheckBalanceCount++;                       // Increment the balance check counter

    String smsNumber = "2424";               // Skinny is the Operator
    String smsMessage2 = "BAL";              // Request the balance of the account

    // Check if it's time to send a balance request
    if (CheckBalanceCount >= 129600) {   // Adjusted check interval (every 5 days) 129600 24hrs
        CheckBalanceCount = 0;                  // Reset counter after sending request

        String command = "+CMGS=\"" + smsNumber + "\"";

        // Clear any leftover data in the modem buffer
        while (SerialAT.available() > 0) {
            SerialAT.read();
            delay(10);  // Ensure buffer clears completely
        }
        delay(100);                                        // Additional delay to ensure buffer is clear added 10/02/2025

        // Send the AT+CMGS command to start the SMS sending process
        modem.sendAT(command);

        // Wait for the '>' prompt from the modem
        unsigned long start = millis();
        bool promptReceived = false;
        while (millis() - start < 5000) {            // Increased timeout to 5000 ms for '>' added 10/02/2025
            if (SerialAT.available()) {
                char c = SerialAT.read();
                if (c == '>') {
                    promptReceived = true;
                    break;
                }
            }
        }

        if (!promptReceived) {  // If '>' prompt not received within timeout
            pDBGln("Error: Modem did not prompt for message input.");
            return;
        }

        // Send the actual SMS message
        SerialAT.print(smsMessage2);

        // Delay slightly before sending the Ctrl+Z (ASCII 26) to finalize the message
        delay(500);  // Brief delay before termination character

        // Send the Ctrl+Z character to complete the SMS
        SerialAT.write(26);

        // Increase wait time after sending the message to handle network delays
        if (modem.waitResponse(10000L) == 1) {  // Increased timeout to 10000 ms for response
            pDBGln("+CMGS Sent balance request!");
        } else {
            pDBGln("Error: Failed to send balance request.");
        }

        // Wait for the balance response SMS
        delay(10000);  // Wait for the response SMS to arrive

        // Read the SMS response
        String response = "";
        while (SerialAT.available()) {
            char c = SerialAT.read();
            response += c;
        }

        // Extract the balance value from the response
        String balance = "";
        bool foundBalance = false;
        for (int i = 0; i < response.length(); i++) {
            // Look for the start of the balance value (digits, '.', or '$')
            if (response[i] == '$' || (response[i] >= '0' && response[i] <= '9')) {
                foundBalance = true;
            }
            // Capture digits and decimal point
            if (foundBalance && (response[i] == '.' || (response[i] >= '0' && response[i] <= '9'))) {
                balance += response[i];
            }
            // Stop capturing when a space or non-numeric character is encountered
            if (foundBalance && (response[i] == ' ' || response[i] == '\r' || response[i] == '\n')) {
                break;
            }
        }

        // Print the extracted balance
        if (balance.length() > 0) {
            pDBGln("Extracted Balance: " + balance);
        } else {
            pDBGln("Error: Could not extract balance from response.");
        }
    }
}`

lewisxhe, 2025-05-06:
```
« AT+IPADDR

+IPADDR: 10.3.196.9
Network IP:10.3.196.9

AT+CMQTTSTART

OK

AT+CCERTDOWN="ca_cert.pem",1970

>-----BEGIN CERTIFICATE-----
MIIFazCCA1OgAwIBAgIRAIIQz7DSQONZRGPgu2OCiwAwDQYJKoZIhvcNAQELBQAw
TzELMAkGA1UEBhMCVVMxKTAnBgNVBAoTIEludGVybmV0IFNlY3VyaXR5IFJlc2Vh
cmNoIEdyb3VwMRUwEwYDVQQDEwxJU1JHIFJvb3QgWDEwHhcNMTUwNjA0MTEwNDM4
WhcNMzUwNjA0MTEwNDM4WjBPMQswCQYDVQQGEwJVUzEpMCcGA1UEChMgSW50ZXJu
ZXQgU2VjdXJpdHkgUmVzZWFyY2ggR3JvdXAxFTATBgNVBAMTDElTUkcgUm9vdCBY
MTCCAiIwDQYJKoZIhvcNAQEBBQADggIPADCCAgoCggIBAK3oJHP0FDfzm54rVygc
h77ct984kIxuPOZXoHj3dcKi/vVqbvYATyjb3miGbESTtrFj/RQSa78f0uoxmyF+
0TM8ukj13Xnfs7j/EvEhmkvBioZxaUpmZmyPfjxwv60pIgbz5MDmgK7iS4+3mX6U
A5/TR5d8mUgjU+g4rk8Kb4Mu0UlXjIB0ttov0DiNewNwIRt18jA8+o+u3dpjq+sW
T8KOEUt+zwvo/7V3LvSye0rgTBIlDHCNAymg4VMk7BPZ7hm/ELNKjD+Jo2FR3qyH
B5T0Y3HsLuJvW5iB4YlcNHlsdu87kGJ55tukmi8mxdAQ4Q7e2RCOFvu396j3x+UC
B5iPNgiV5+I3lg02dZ77DnKxHZu8A/lJBdiB3QW0KtZB6awBdpUKD9jf1b0SHzUv
KBds0pjBqAlkd25HN7rOrFleaJ1/ctaJxQZBKT5ZPt0m9STJEadao0xAH0ahmbWn
OlFuhjuefXKnEgV4We0+UXgVCwOPjdAvBbI+e0ocS3MFEvzG6uBQE3xDk3SzynTn
jh8BCNAw1FtxNrQHusEwMFxIt4I7mKZ9YIqioymCzLq9gwQbooMDQaHWBfEbwrbw
qHyGO0aoSCqI3Haadr8faqU9GY/rOPNk3sgrDQoo//fb4hVC1CLQJ13hef4Y53CI
rU7m2Ys6xt0nUW7/vGT1M0NPAgMBAAGjQjBAMA4GA1UdDwEB/wQEAwIBBjAPBgNV
HRMBAf8EBTADAQH/MB0GA1UdDgQWBBR5tFnme7bl5AFzgAiIyBpY9umbbjANBgkq
hkiG9w0BAQsFAAOCAgEAVR9YqbyyqFDQDLHYGmkgJykIrGF1XIpu+ILlaS/V9lZL
ubhzEFnTIZd+50xx+7LSYK05qAvqFyFWhfFQDlnrzuBZ6brJFe+GnY+EgPbk6ZGQ
3BebYhtF8GaV0nxvwuo77x/Py9auJ/GpsMiu/X1+mvoiBOv/2X/qkSsisRcOj/KK
NFtY2PwByVS5uCbMiogziUwthDyC3+6WVwW6LLv3xLfHTjuCvjHIInNzktHCgKQ5
ORAzI4JMPJ+GslWYHb4phowim57iaztXOoJwTdwJx4nLCgdNbOhdjsnvzqvHu7Ur
TkXWStAmzOVyyghqpZXjFaH3pO3JLF+l+/+sKAIuvtd7u+Nxe5AW0wdeRlN8NwdC
jNPElpzVmbUq4JUagEiuTDkHzsxHpFKVK7q4+63SM1N95R1NbdWhscdCb+ZAJzVc
oyi3B43njTOQ5yOf+1CceWxG1bQVs5ZufpsMljq4Ui0/1lvh+wjChP4kqKOJ2qxq
4RgqsahDYVvTH9w7jXbyLeiNdd8XM2w9U/t7y0Ff/9yi0GE44Za4rF2LN9d11TPA
mRGunUHBcnWEvgJBQl9nJEiU0Zsnvgc/ubhPgXRR4Xq37Z0j4r7g1SgEEzwxA57d
emyPxgcYxn/eR44/KJ4EBs+lVDR3veyJm+kXQ99b21/+jh5Xos1AnX5iItreGCc=
-----END CERTIFICATE-----

OK
AT+CSSLCFG="cacert",0,"ca_cert.pem"

OK
AT+CSSLCFG="sslversion",0,4

OK
AT+CMQTTSSLCFG=0,0

OK
AT+CSSLCFG="enableSNI",0,1

OK
AT+CSSLCFG="authmode",0,1

OK
AT+CMQTTREL=0

OK
AT+CMQTTACCQ=0,"SIM70XXX",1

OK
AT+CMQTTCFG="version",0,4

OK
AT+CMQTTCONNECT=0,"tcp://xxxxxxxxxx.s2.eu.hivemq.cloud:8883",60,1,"SIM70XXX","Password"

OK

+CMQTTCONNECT: 0,0
AT+CMQTTDISC?

+CMQTTDISC: 0,0

+CMQTTDISC: 1,1

OK

AT+CMQTTSUB=0,21,0,0

>GsmMqttTest/subscribe

OK

+CMQTTSUB: 0,0
AT+SIMCOMATI

Manufacturer: SIMCOM INCORPORATED
Model: SIMCOM_SIM7600G-H
Revision: LE20B04SIM7600G22
QCN: 
IMEI: xxxxxxxxxxxx
MEID: 
+GCAP: +CGSM
DeviceInfo: 173,170

OK
AT+CMQTTTOPIC=0,19

>GsmMqttTest/publish

OK
AT+CMQTTPAYLOAD=0,10

>RunTime:11

OK
AT+CMQTTPUB=0,0,60

+CMQTTPUB: 0,0

OK
» AT+CMQTTDISC?
« 
+CMQTTDISC: 0,0

+CMQTTDISC: 1,1

OK
» AT+CMQTTDISC?
« 
+CMQTTDISC: 0,0

+CMQTTDISC: 1,1

OK
» AT+CMQTTDISC?
« 
+CMQTTDISC: 0,0

+CMQTTDISC: 1,1

OK
» AT+CMQTTDISC?
« 
+CMQTTDISC: 0,0

+CMQTTDISC: 1,1

OK
» AT+CMGS="+xxxxxxxxxx"
« 
> 
» HELLO
« 
> 
» 1a 0D 0A
« 

+CMGS: 70

OK
» AT+CMQTTDISC?
« 
+CMQTTDISC: 0,0

+CMQTTDISC: 1,1

OK
» AT+CMQTTDISC?
« 
+CMQTTDISC: 0,0

+CMQTTDISC: 1,1

OK
» AT+CMQTTDISC=0,120            -----------Manually disconnect
« 
+CMQTTDISC: 0,0

OK
» AT+CMQTTDISC?
« 
+CMQTTDISC: 0,1

+CMQTTDISC: 1,1

OK
» AT+CMQTTDISC?
« 
+CMQTTDISC: 0,1

+CMQTTDISC: 1,1

OK

```
Please upgrade the latest firmware for testing. I tested the same PN as you and did not get this problem. Of course, I used the built-in MQTT for testing.

droidblastnz, 2025-05-06:
@lewisxhe  I am using 1883 and the issue is switching between a MQTT data connection and then SMS/Cellular. My question was around the steps to disconnect the data connection, receive and read a sms re-connection data/MQTT and publish. 

Its not a question of MQTT or MQTTS its the steps from data to cellular back to a data connection please :)

I see a few user with the exact same issue.

If you could detail the disconnect steps and reconnect steps please?

droidblastnz, 2025-05-06:
> What I demonstrated is connecting to MQTT (regardless of the port) and then performing SMS operations?
> 
> Don't you mean that when there is a data task currently in a connected state, the SMS operation is performed (the network may be switched during this period), and then the data is disconnected, right?

Correct when you have a data connection then try and send cellular as in a txt or call it will disconnect data take the sms or call and sometimes reconnect. In my case the SIM7600 is remote (4 hrs away) and goes offline. 

What is the process to code. For example AT+NETCLOSE after CIPCLOSE..

If connect by data do you need to close the connection/socket to use cellular? 
Then to start data do you AT+NETOPEN?

It seems to work then sooner or later you need a hard reboot to recover?

lewisxhe, 2025-05-06:
Normally, there is no need to actively intervene in the network and SMS services. In theory, it will switch automatically.

When the network cannot be connected, have you checked AT+NETOPEN?

Is it in the open state?

droidblastnz, 2025-05-06:
Logs from bad then good  I dont see - AT+NETOPEN?

Bad

```
SMS: Skinny Balance Check
AT+CMGR=0
message: 
+CMGR: "REC READ","2424","","25/04/10,18:45:46+48"
Hi! Your Skinny balance is
1.65 credit
For your account info, text INFO to 2424.

OK

Get balance: 2424.00
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CIPSEND=0,31

>0 topic/balance2424.00
OK

+CIPSEND: 0,31,31
AT+CMGD=0

OK
SMS: Checking for unauthorized messages...
AT+CMGL="ALL"

OK
SMS: Deleting all messages...
AT+CMGD=0,4

OK
AT+CGREG?

+CGREG: 0,1

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CGREG?

+CGREG: 0,1

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CGREG?

+CGREG: 0,1

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CGREG?

+CGREG: 0,1

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CGREG?

+CGREG: 0,1

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CGREG?

+CGREG: 0,1

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CGREG?

+CGREG: 0,1

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0
```

Good 

```
OK
Checking for incoming SMS messages...
AT+CMGL="REC UNREAD"
AT+CMGR=0
Message content: 
+CMGR: "REC READ","2424","","24/12/11,03:36:44+52"
Hi! Your Skinny balance is
$14.65 credit
For your account info, text INFO to 2424.

OK

SMS: Skinny Balance Check
AT+CMGR=0
message: 
+CMGR: "REC READ","2424","","24/12/11,03:36:44+52"
Hi! Your Skinny balance is
$14.65 credit
For your account info, text INFO to 2424.

OK

Get balance: 14.65
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CIPSEND=0,29

>0 topic/balance14.65
OK

+CIPSEND: 0,29,29
AT+CMGD=0

OK
SMS: Checking for unauthorized messages...
AT+CMGL="ALL"

OK
SMS: Deleting all messages...
AT+CMGD=0,4

OK
AT+CGREG?

+CGREG: 0,1

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CGREG?

+CGREG: 0,1

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CGREG?

+CGREG: 0,1

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CGREG?

+CGREG: 0,1

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CGREG?

+CGREG: 0,1

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CGREG?

+CGREG: 0,1

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CGREG?

+CGREG: 0,1

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0
```

lewisxhe, 2025-05-06:
😂 Is AT+NETOPEN? included in your code? If you only use the MqttClient example, then AT+NETOPEN will only be enabled when connecting to the network, and then it will never check the network status. You should add AT+NETOPEN? to your code to send the network check command regularly. Do you understand my explanation?

droidblastnz, 2025-05-06:
> AT+NETOPEN  my code is mainly [this- AllFunctions](https://github.com/Xinyuan-LilyGO/T-SIM7600X/blob/master/examples/AllFunctions/AllFunctions.ino) so not I dont see AT+NETOPEN being used?

So the AllFunctions example has the #define TINY_GSM_MODEM_SIM7600 enabled so this should call the sendAT(GF("+NETOPEN"));?

Of should I add sendAT(GF("+NETOPEN")); post checking for SMS?

 ```
   // Start the socket service

    // This activates and attaches to the external PDP context that is tied
    // to the embedded context for TCP/IP (ie AT+CGACT=1,1 and AT+CGATT=1)
    // Response may be an immediate "OK" followed later by "+NETOPEN: 0".
    // We to ignore any immediate response and wait for the
    // URC to show it's really connected.
    sendAT(GF("+NETOPEN"));
    if (waitResponse(75000L, GF(AT_NL "+NETOPEN: 0")) != 1) { return false; }

    return true;
  }

  bool gprsDisconnectImpl() {
    // Close all sockets and stop the socket service
    // Note: On the LTE models, this single command closes all sockets and the
    // service
    sendAT(GF("+NETCLOSE"));
    if (waitResponse(60000L, GF(AT_NL "+NETCLOSE: 0")) != 1) { return false; }

    return true;
  }

  bool isGprsConnectedImpl() {
    sendAT(GF("+NETOPEN?"));
    // May return +NETOPEN: 1, 0.  We just confirm that the first number is 1
    if (waitResponse(GF(AT_NL "+NETOPEN: 1")) != 1) { return false; }
    waitResponse();

    sendAT(GF("+IPADDR"));  // Inquire Socket PDP address
    // sendAT(GF("+CGPADDR=1")); // Show PDP address
    if (waitResponse() != 1) { return false; }

    return true;
  }
```

droidblastnz, 2025-05-06:
Connected to GPRS/Data/MQTT - logs below is a excerpt when I see the issue about to occur. Send BAL to 2424 to get remaining credit balance and it returned 3.35 (technically it returns $3.35 but you see a control character `STX` in place of the $). 

If is there meant to pub to MQTT topic/balance3.35 but we see topic/balance2424.00 published.
At this point the modem will shortly go off line and no longer send or receive MQTT commands. 

I have checked and rewritten the credit balance checks code block and can confirm this works well and as expected until I see the output below.  So I request a credit balance every 36 hrs and want to publish via MQTT.

Serial logs

```
09:46:29.544 -> OK
09:46:29.544 -> Checking for incoming SMS messages...
09:46:29.544 -> AT+CMGL="REC UNREAD"
09:46:30.576 -> AT+CMGR=0
09:46:31.705 -> Message content: 
09:46:31.705 -> +CMGR: "REC READ","2424","","25/04/05,09:45:52+52"
09:46:31.705 -> Hi! Your Skinny balance is
09:46:31.705 -> 3.35 credit
09:46:31.705 -> For your account info, text INFO to 2424.
09:46:31.705 -> 
09:46:31.705 -> OK
09:46:31.705 -> 
09:46:31.705 -> SMS: Skinny Balance Check
09:46:31.705 -> AT+CMGR=0
09:46:32.736 -> message: 
09:46:32.736 -> +CMGR: "REC READ","2424","","25/04/05,09:45:52+52"
09:46:32.736 -> Hi! Your Skinny balance is
09:46:32.736 -> 3.35 credit
09:46:32.736 -> For your account info, text INFO to 2424.
09:46:32.736 -> 
09:46:32.736 -> OK
09:46:32.736 -> 
09:46:32.736 -> Get balance: 2424.00
09:46:32.736 -> AT+CIPRXGET=4,0
09:46:32.736 -> 
09:46:32.796 -> +CIPRXGET: 4,0,0
09:46:32.796 -> 
09:46:32.796 -> OK
09:46:32.796 -> AT+CIPCLOSE?
09:46:32.796 -> 
09:46:32.796 -> +CIPCLOSE: 1,0,0,0,0,0,0,0,0,0
09:46:32.841 -> 
09:46:32.841 -> OK
09:46:32.841 -> AT+CIPSEND=0,31
09:46:32.841 -> 
09:46:32.888 -> >0
09:46:32.888 -> >topic/balance2424.00
09:46:32.888 -> >0
```

Putty Logs

```
OK
Checking for incoming SMS messages...
AT+CMGL="REC UNREAD"
AT+CMGR=0
Message content: 
+CMGR: "REC READ","2424","","25/04/05,09:52:53+52"
Hi! Your Skinny balance is
3.35 credit
For your account info, text INFO to 2424.

OK

SMS: Skinny Balance Check
AT+CMGR=0
message: 
+CMGR: "REC READ","2424","","25/04/05,09:52:53+52"
Hi! Your Skinny balance is
3.35 credit
For your account info, text INFO to 2424.

OK

Get balance: 2424.00
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0

OK
AT+CIPSEND=0,31

>0 topic/balance2424.00    (control characters GS and DC4 are seen here)
OK

+CIPSEND: 0,31,31
AT+CMGD=0

OK
SMS: Checking for unauthorized messages...
AT+CMGL="ALL"

OK
SMS: Deleting all messages...
AT+CMGD=0,4
```

lewisxhe, 2025-05-07:
```

[15:19:59.934]-> AT+CMQTTDISC?

[15:19:59.960]<- AT+CMQTTDISC?
+CMQTTDISC: 0,0
+CMQTTDISC: 1,1

OK

[15:20:30.054]-> AT+CMQTTDISC?

[15:20:30.086]<- AT+CMQTTDISC?
+CMQTTDISC: 0,0
+CMQTTDISC: 1,1

OK

[15:21:54.165]-> 
[15:21:56.797]-> AT+CMGS="+86xxxxxxxxxxxxx"

[15:21:56.837]<- AT+CMGS="+86xxxxxxxxxxxxx"
> 
[15:22:06.989]-> 15:22

[15:22:07.007]<- 
> 15:22
[15:22:08.501]-> 
[15:22:08.513]<- 

[15:22:08.914]<- 
+CMGS: 71

OK

[15:22:18.926]-> AT+CMQTTDISC?

[15:22:18.955]<- AT+CMQTTDISC?
+CMQTTDISC: 0,0
+CMQTTDISC: 1,1

OK

[15:22:46.052]<- 
+CMTI: "SM",27

[15:22:48.126]-> AT+CMGL="REC UNREAD"

[15:22:48.159]<- AT+CMGL="REC UNREAD"
[15:22:48.260]<- 
+CMGL: 27,"REC UNREAD","xxxxxxxxxxxxx","","25/05/07,15:22:46+32"
send to sim7600

OK

[15:22:51.821]-> AT+CMGR=0

[15:22:51.843]<- AT+CMGR=0
[15:22:51.942]<- 
+CMGR: "REC READ","10000","","22/06/14,16:00:40+32"
51715EFA7F8E4E3D5BB656ED3002

OK

[15:22:56.118]-> AT+CMQTTDISC?

[15:22:56.145]<- AT+CMQTTDISC?
+CMQTTDISC: 0,0
+CMQTTDISC: 1,1

OK

[15:22:57.373]-> AT+CMQTTDISC?

[15:22:57.402]<- AT+CMQTTDISC?
+CMQTTDISC: 0,0
+CMQTTDISC: 1,1

OK

[15:30:13.718]-> AT+CMQTTDISC?

[15:30:13.746]<- AT+CMQTTDISC?
+CMQTTDISC: 0,0
+CMQTTDISC: 1,1

OK

[15:30:14.502]-> AT+CMQTTDISC?

[15:30:14.536]<- AT+CMQTTDISC?
+CMQTTDISC: 0,0
+CMQTTDISC: 1,1

OK

[15:30:16.350]-> AT+CMGL="REC UNREAD"

[15:30:16.386]<- AT+CMGL="REC UNREAD"
OK

[15:32:41.822]-> AT+CMQTTDISC?

[15:32:41.854]<- AT+CMQTTDISC?
+CMQTTDISC: 0,0
+CMQTTDISC: 1,1

OK

[15:55:20.694]-> AT+CMQTTDISC?

[15:55:20.729]<- AT+CMQTTDISC?
+CMQTTDISC: 0,0
+CMQTTDISC: 1,1

OK

[16:28:16.112]-> AT+CMQTTDISC?

[16:28:16.139]<- AT+CMQTTDISC?
+CMQTTDISC: 0,0
+CMQTTDISC: 1,1

OK

[16:59:37.361]-> AT+CMQTTDISC?

[16:59:37.391]<- AT+CMQTTDISC?
+CMQTTDISC: 0,0
+CMQTTDISC: 1,1

OK

[18:06:29.950]-> AT+CMQTTDISC?

[18:06:29.982]<- AT+CMQTTDISC?
+CMQTTDISC: 0,0
+CMQTTDISC: 1,1

OK

[18:06:30.637]-> AT+CMQTTDISC?

[18:06:30.669]<- AT+CMQTTDISC?
+CMQTTDISC: 0,0
+CMQTTDISC: 1,1

OK

[18:08:01.059]<- 
+CMTI: "SM",28

[18:08:03.286]-> AT+CMGL="REC UNREAD"

[18:08:03.320]<- AT+CMGL="REC UNREAD"
[18:08:03.419]<- 
+CMGL: 28,"REC UNREAD","xxxxxxxxxxxxx","","25/05/07,18:08:00+32"
test

OK

[18:08:05.269]-> AT+CMGR=0

[18:08:05.290]<- AT+CMGR=0
[18:08:05.390]<- 
+CMGR: "REC READ","10000","","22/06/14,16:00:40+32"
51715EFA7F8E4E3D5BB656ED3002

OK

```

I tested it for a few hours and it didn't disconnect.

droidblastnz, 2025-05-07:
> I need you to capture the modem log so that SIMCOM can analyze the disconnection behavior.
> 
> Since the USB-C of this SIM7600G board is very close, only one USB-C can be connected at the same time. The capture log must be connected to the USB interface of the MODEM to use it. At this time, the USB2TLL interface of ESP32 cannot be used. You need to use an external USB2TTL to connect to ESP32 RX TX to monitor the behavior of the modem. After accurately capturing the data disconnection, export the log and send it to me, and I will send it to SIMCOM for analysis.
> 
> [capture tool](https://drive.google.com/file/d/11x_J4RiHb168PCx9OwyLoqq1MEj0l18P/view?usp=sharing)

Thanks SIM7600 is in a remote town location so I can do this very quickly but thanks for assisting. 

I note in its current state that I cannot even ring the Sim7600 which means the modem is hung and no MQTT commands or SMS will work in its current state.

When I do get to the SIM7600 I will check LEDs. Note mine is power with a buck convertor not solar.

    LED_G7 yellow If you see it lights up, it means the battery is connected in reverse, it will not light up when the battery is connected normally
    LED_G6 green Charging completion indicator, lights up after charging is completed
    LED_G3 red Charging indicator, lights up when charging is normal
    LED_G5 red Network status indicator
    LED_G8 red Modem status indicator, it will light up when the device is turned on normally