Xinyuan-LilyGO/T-SIM7600X #131: sql

norielcahilig, 2025-05-26:
good day sir, do you have a program template/ example for us to connect t-sim7600G to a database(mysql)?

lewisxhe, 2025-05-26:
This may require API support from the website, which I do not have.

droidblastnz, 2025-05-28:
Or API.

```
#include <SoftwareSerial.h>
SoftwareSerial sim7600(10, 11); // RX, TX

void setup() {
  Serial.begin(115200);
  sim7600.begin(9600);

  sendAT("AT", "OK");
  sendAT("AT+CPIN?", "READY");
  sendAT("AT+CGATT?", "1");
  sendAT("AT+CSTT=\"your.apn.here\"", "OK");
  sendAT("AT+CIICR", "OK");
  sendAT("AT+CIFSR");  // Get IP
  sendAT("AT+HTTPINIT", "OK");
  sendAT("AT+HTTPPARA=\"CID\",1", "OK");
  sendAT("AT+HTTPPARA=\"URL\",\"http://yourserver.com/api/submit\"", "OK");
  sendAT("AT+HTTPPARA=\"CONTENT\",\"application/x-www-form-urlencoded\"", "OK");

  // POST data
  sendAT("AT+HTTPDATA=32,10000", "DOWNLOAD");
  sim7600.print("device_id=123&temp=25.4");
  delay(1000);
  sendAT("AT+HTTPACTION=1", "200"); // POST
  sendAT("AT+HTTPTERM", "OK");
}

void loop() {}

bool sendAT(String cmd, String expected, int timeout = 3000) {
  sim7600.println(cmd);
  long start = millis();
  String resp = "";
  while (millis() - start < timeout) {
    while (sim7600.available()) {
      resp += (char)sim7600.read();
    }
    if (resp.indexOf(expected) != -1) {
      Serial.println("✅ " + cmd);
      return true;
    }
  }
  Serial.println("❌ " + cmd + " => " + resp);
  return false;
}
```

pip install flask mysql-connector-python

api_server.py

```
from flask import Flask, request
import mysql.connector

app = Flask(__name__)

@app.route('/api/submit', methods=['POST'])
def submit():
    device_id = request.form.get('device_id')
    temp = request.form.get('temp')
    if not device_id or not temp:
        return "Missing data", 400

    conn = mysql.connector.connect(
        host="localhost",
        user="your_user",
        password="your_pass",
        database="your_db"
    )
    cursor = conn.cursor()
    cursor.execute("INSERT INTO sensor_data (device_id, temperature) VALUES (%s, %s)", (device_id, temp))
    conn.commit()
    cursor.close()
    conn.close()

    return "Data received", 200

if __name__ == '__main__':
    app.run(host="0.0.0.0", port=80)
```

SQL

```
CREATE TABLE sensor_data (
  id INT AUTO_INCREMENT PRIMARY KEY,
  device_id VARCHAR(20),
  temperature FLOAT,
  timestamp TIMESTAMP DEFAULT CURRENT_TIMESTAMP
);
```

droidblastnz, 2025-05-29:
Try - SMS with link ... I havent tried so see how you go.

```
#include <SoftwareSerial.h>

SoftwareSerial sim7600(7, 8); // RX, TX (adjust pins as needed)

void setup() {
  Serial.begin(115200);
  sim7600.begin(9600); // Your SIM7600 baud rate

  delay(3000);
  sendSMS("+1234567890", "Check this out: https://example.com");
}

void loop() {
  // Nothing here
}

void sendSMS(const char* phoneNumber, const char* message) {
  sim7600.println("AT");
  waitForResponse();

  sim7600.println("AT+CMGF=1"); // Set SMS to text mode
  waitForResponse();

  sim7600.print("AT+CMGS=\"");
  sim7600.print(phoneNumber);
  sim7600.println("\"");
  delay(1000);

  sim7600.print(message);
  delay(100);
  sim7600.write(26); // CTRL+Z to send
  waitForResponse(10000); // Wait longer for final response
}

void waitForResponse(unsigned long timeout = 2000) {
  unsigned long start = millis();
  while (millis() - start < timeout) {
    while (sim7600.available()) {
      Serial.write(sim7600.read());
    }
  }
}
```

droidblastnz, 2025-05-29:
> yea, i think all network providers here in our country are required to do
> so.
> […](#)
> 
>  <[#131 (comment)](https://github.com/Xinyuan-LilyGO/T-SIM7600X/issues/131#issuecomment-2920450207)>
> 
>  we actually doing this, but it's actually our network provider blocking sms
>  that contains link
>  … <#m_-4987924517377576474_>
> 
>  <[#131](https://github.com/Xinyuan-LilyGO/T-SIM7600X/issues/131)>)
>  <[#131](https://github.com/Xinyuan-LilyGO/T-SIM7600X/issues/131) (comment)
>  <[#131 (comment)](https://github.com/Xinyuan-LilyGO/T-SIM7600X/issues/131#issuecomment-2917882425)>
>  >
> 
>  Try - SMS with link ... I havent tried so see how you go.
> 
>  #include <SoftwareSerial.h>
> 
>  SoftwareSerial sim7600(7, 8); // RX, TX (adjust pins as needed)
> 
>  void setup() {
>  Serial.begin(115200);
>  sim7600.begin(9600); // Your SIM7600 baud rate
> 
>  delay(3000);
>  sendSMS("+1234567890", "Check this out: https://example.com");
>  }
> 
>  void loop() {
>  // Nothing here
>  }
> 
>  void sendSMS(const char* phoneNumber, const char* message) {
>  sim7600.println("AT");
>  waitForResponse();
> 
>  sim7600.println("AT+CMGF=1"); // Set SMS to text mode
>  waitForResponse();
> 
>  sim7600.print("AT+CMGS="");
>  sim7600.print(phoneNumber);
>  sim7600.println(""");
>  delay(1000);
> 
>  sim7600.print(message);
>  delay(100);
>  sim7600.write(26); // CTRL+Z to send
>  waitForResponse(10000); // Wait longer for final response
>  }
> 
>  void waitForResponse(unsigned long timeout = 2000) {
>  unsigned long start = millis();
>  while (millis() - start < timeout) {
>  while (sim7600.available()) {
>  Serial.write(sim7600.read());
>  }
>  }
>  }
> 
>  <[#131](https://github.com/Xinyuan-LilyGO/T-SIM7600X/issues/131) (comment)
>  <[#131 (comment)](https://github.com/Xinyuan-LilyGO/T-SIM7600X/issues/131#issuecomment-2917882425)>
>  >,
> 
>  *@*.***>
> 
>  Are they blocking for security? Can you try another provider?
> 
>  <[#131 (comment)](https://github.com/Xinyuan-LilyGO/T-SIM7600X/issues/131#issuecomment-2920450207)>,

Providers are? and what is the error?

norielcahilig, 2025-05-29:
globe, smart, other telcos here in Philippines

droidblastnz, 2025-05-29:
> globe, smart, other telcos here in Philippines
> […](#)

Instead of `https://example.com` try [example.com](url)

droidblastnz, 2025-05-29:
Let me know the exact message/link please

droidblastnz, 2025-05-29:
> User : [User's name] Location: [Longitude, Latitude] Map: ***@***.***

Assume
```
User: John Doe
Location: 14.5995, 120.9842

Map: maps.example.com/user123
(Please copy and paste into your browser)
```
Try 

```
User: John Doe
Location: 14.5995, 120.9842

Map: maps.example.com/user123
(Please copy and paste into your browser)
```

See if that works and doesnt trigger the providers security.

droidblastnz, 2025-05-29:
Or 

```
User: John Doe  
Location: 14.5995, 120.9842  

Map: google(dot)com/maps?q=14.5995,120.9842  
(copy & paste into browser)
```

C++

```
String userName = "John Doe";
float latitude = 14.5995;
float longitude = 120.9842;

String message = "User: " + userName + "\n";
message += "Location: " + String(latitude, 4) + ", " + String(longitude, 4) + "\n";
message += "Map: google(dot)com/maps?q=" + String(latitude, 4) + "," + String(longitude, 4);

sendSMS("+639XXXXXXXXX", message.c_str());
```

norielcahilig, 2025-05-29:
message was sent, but still not visible on receiver's phone

droidblastnz, 2025-05-30:
> message was sent, but still not visible on receiver's phone
> […](#)

Work you way thru the code.

`sendSMS("+639XXXXXXXXX", "Hello from SIM7600");`

```
AT
AT+CMGF=1
AT+CSCS="GSM"
AT+CMGS="+639XXXXXXXXX"
> Hello test (press Ctrl+Z or send ASCII 26)
```

Serial Logs do you see

```
+CMGS: <msg_id> → Message accepted
OK → Sent
```
Or errors like +CMS ERROR: 500?

try delays

```
sim7600.println("AT");
delay(500);
sim7600.println("AT+CMGF=1");
delay(500);
```

norielcahilig, 2025-05-30:
we do not have any errors regarding these +CMSs, infact we are successfully
sending sms BUT without links, and we must have sms with clickable links
for ease of use of the user

droidblastnz, 2025-05-30:
> we do not have any errors regarding these +CMSs, infact we are successfully
> sending sms BUT without links, and we must have sms with clickable links
> for ease of use of the user
> […](#)

What works?

- http://www.domain.com
- https://www.domain.com
- www.domain.com
- domain.com
- bit.ly
- t.me

What is the provider picking up on?

If the provider (assume Smart and Globe) is blocking not much you can do apart from work what triggers.

norielcahilig, 2025-05-30:
where we could retrieve 't.me' link? we have tried every one of these,
except from t.me link, i assume it would not also work, but we'll give it a
try

droidblastnz, 2025-05-30:
https://www.reddit.com/r/Philippines/comments/xrtwwc/globe_starts_blocking_sms_with_clickable_links/

_Funny though after globe implemented this I [receive](https://i.imgur.com/BbVC17f.png) another spam message again but this time they remove the link/s and replaced dot into comma. Kumag talaga 'tong mga to._

norielcahilig, 2025-05-30:
yea hahahaha scam senders actually make their way through my sms app

droidblastnz, 2025-05-30:
> yea hahahaha scam senders actually make their way through my sms app
> […](#)

[](https://www.reddit.com/r/Tech_Philippines/comments/1dtgpj8/iphone_to_android_textsms_cannot_send_links_email/)

Dont think you will get a workaround with out adding spacing.  They may use regex to check for the https://, http: or @ or .domain.com

norielcahilig, 2025-05-31:
hi, do u know if how we could use lithium-ion battery(the flat ones) to the
board, what pins are we going to connect it? thank you