Xinyuan-LilyGO/T-SIM7600X #133: gmail

norielcahilig, 2025-06-02:
do u have any example that makes the board send emails? thank you

lewisxhe, 2025-06-03:
There are no examples for this

norielcahilig, 2025-06-04:
> #include <SoftwareSerial.h>
> SoftwareSerial sim7600(26, 27); // RX, TX (adjust to your wiring)
>
> const char* smtp_server = "mail.smtp2go.com";
> const int smtp_port = 2525;
> const char* smtp_user = "your_smtp2go_username";
> const char* smtp_pass = "your_smtp2go_password";
>
> void sendEmail() {
>   sim7600.println("AT+CSQ"); delay(500);
>   sim7600.println("AT+CGATT=1"); delay(1000);
>   sim7600.println("AT+CIPSHUT"); delay(1000);
>   sim7600.println("AT+CIPMUX=0"); delay(500);
>   sim7600.println("AT+CSTT=\"wapaccess.co.nz\""); delay(1000);
>   sim7600.println("AT+CIICR"); delay(3000);
>   sim7600.println("AT+CIFSR"); delay(1000);
>
>   sim7600.println("AT+CIPSTART=\"TCP\",\"mail.smtp2go.com\",\"2525\"");
>   delay(5000); // Wait for connection
>
>   sim7600.println("AT+CIPSEND");
>   delay(2000);
>
>   // SMTP handshake and email body
>   sim7600.print("HELO smtp2go.com\r\n");
>   delay(500);
>   sim7600.print("AUTH LOGIN\r\n"); delay(500);
>   sim7600.print("your_base64_encoded_username\r\n"); delay(500);
>   sim7600.print("your_base64_encoded_password\r\n"); delay(500);
>   sim7600.print("MAIL ***@***.***>\r\n"); delay(500);
>   sim7600.print("RCPT ***@***.***>\r\n"); delay(500);
>   sim7600.print("DATA\r\n"); delay(500);
>   sim7600.print("Subject: Test from SIM7600\r\n");
>   sim7600.print("\r\nThis is a test email from the SIM7600.\r\n.\r\n");
>   delay(500);
>   sim7600.print("QUIT\r\n");
>
>   // End the send
>   delay(500);
>   sim7600.write(0x1A); // CTRL+Z to send
>   delay(3000);
> }
>
> String encodeBase64(String str) {
>   // Optional: Add your own Base64 encoding function here or pre-encode online.
>   return "Base64String";
> }

does this need wifi? or we can use sim7600 cellular data?