Xinyuan-LilyGO/T-SIM7600X #134: 7600G-H R2 GPS Issue

castironclay, 2025-06-14:
I have seen other threads on here related to GPS issues with the 7600G-H R2 and have gone through them but not having any luck with fixing my issue. I am booting the modem and enabling GPS via this sketch.
```c
void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  delay(1000);
  Serial.println("Starting HTTPS Webhook Test with Interactive Mode...");

  SerialAT.begin(SIM7600_BAUD, SERIAL_8N1, RX_PIN, TX_PIN);
  delay(2000);

  // Power on modem
  Serial.println("Powering on modem...");
  pinMode(PWR_PIN, OUTPUT);
  digitalWrite(PWR_PIN, HIGH);  // Inactive
  delay(500);
  digitalWrite(PWR_PIN, LOW);  // Power on
  delay(1000);
  digitalWrite(PWR_PIN, HIGH);  // Release
  delay(10000);                 // Wait for boot

  // Initialize modem
  if (!modem.init()) {
    Serial.println("Failed to initialize modem! Restarting");
    modem.restart();
  }
  Serial.print("Modem Info: ");
  Serial.println(modem.getModemInfo());
  modem.enableGPS();
  delay(3000);
}

void loop() {
  // Passthrough: Serial Monitor to modem
  if (Serial.available()) {
    String input = Serial.readStringUntil('\n');
    input.trim();
    if (input.length() > 0) {
      Serial.println("Sending: " + input);
      SerialAT.println(input);
    }
  }

  // Relay modem responses
  if (SerialAT.available()) {
    Serial.println("Modem: " + SerialAT.readString());
  }
}
```
GPS status
```bash
Sending: at+cgps?
Modem: 
+CGPS: 1,1
OK

Sending: at+cgpsauto?
Modem: 
+CGPSAUTO: 1
OK

```
The issue I'm having is the output from a `AT+CGPSINFO` command will show me location information very quickly after booting (happy about that) but it never updates even when I'm moving. I'm new to using this platform and AT commands in general so any help would be appreciated. Modem info below. 
```bash
Modem: 
Manufacturer: SIMCOM INCORPORATED
Model: SIMCOM_SIM7600G-H
Revision: LE20B05SIM7600G22
QCN: 
IMEI: --redacted--
MEID: 
+GCAP: +CGSM
DeviceInfo: 173,170
```

lewisxhe, 2025-06-16:
You said that even if you move the location, the GPS data will not be updated? ?
How much distance do you move? ?

https://github.com/Xinyuan-LilyGO/T-SIM7600X/blob/a2c0a34430fa93f9a518b379e282106ff1c5b636/examples/GPS_SMS/GPS_SMS.ino#L84

There is also a flight mode that you did not set.

lewisxhe, 2025-06-17:
It seems that it is not possible. Airplane mode will turn off all radio frequencies.

lewisxhe, 2025-06-18:
Set GPIO25 to HIGH to disable airplane mode, otherwise the device is offline

castironclay, 2025-06-18:
You’re saying I should do this to ensure the device is not in airplane mode? I don’t think it’s in airplane mode due to me being able to send HTTP requests. In general it’s working well other than after that initial gps sync it doesn’t seem to update its location. I check at+cgpsinfo as I move around and it’s always the same.

lewisxhe, 2025-06-19:
What is the approximate difference in your moving distance?

castironclay, 2025-06-19:
After the initial power on and gpsinfo check I have moved hundreds of meters. Could you at least confirm this should work? I want to make sure I’m not dealing with a known issue.

castironclay, 2025-06-19:
I have done testing by powering the device with my laptop and walking up and down the sidewalk while watching the Arduino serial monitor. This is where I saw the gpsinfo output not changing.

lewisxhe, 2025-06-21:
Short distances will not produce noticeable changes, you must move over long distances to observe the problem