Xinyuan-LilyGO/T-SIM7600X #136: Can't connect to LTE 4G  in France with T-SIM7600E-L1C  firmware SIM7600M21-A_V2.0.1

snowgato, 2025-06-28:
Hi,
I have bought on aliexpress the lilygo T-SIM7600E-L1C   [lilygo T-SIM7600E-L1C](https://fr.aliexpress.com/item/1005003154792763.html?spm=a2g0o.order_list.order_list_main.35.f0755e5b1porBK&gatewayAdapt=glo2fra)
It is supposed according to the specifications on the merchant web page to work on B1/B3/B5/B7/B8/B20 bands, thus I should be able to connect on SFR ( french telecom broadcaster) 4G network.
Unfortunately, the module doesnt connect to 4G with AT commands whenI insert a sim card with 4G access and PIN code deactivated
I have inserted the 4G antenna on "main" connection

AT+CNMP=13
AT+CNMP=13
OK
AT+COPS?
AT+COPS?
+COPS: 1,0,"F SFR",0
OK

ATI
ATI
Manufacturer: SIMCOM INCORPORATED
Model: SIMCOM_SIM7600E
Revision: SIM7600M21-A_V2.0.1
IMEI: 861005079131568
+GCAP: +CGSM

Deepseek tells me this firmware ( SIM7600M21-A_V2.0.1) is related to a module meant to work in america and doesnt support any of the following bands B1 / B3 / B7 / B20 / B28 and thus there's no way I can get 4G connexion with that module.
Does someone know something about this module :SIM7600M21-A_V2.0.1 and  what can be done ?
I have the impression that I have to send it back
Thanks for your help

snowgato, 2025-06-28:
Hi,
Thanks for your help !
Here are the results :

AT+CPIN?
AT+CPIN?
+CPIN: READY
OK

AT+CSQ
AT+CSQ
+CSQ: 27,99
OK

AT+CGREG?
AT+CGREG?
+CGREG: 0,1
OK

AT+COPS?
AT+COPS?
+COPS: 1,0,"F SFR",0
OK

AT+CPSI?
AT+CPSI?
+CPSI: GSM,Online,208-10,0xb5ab,11520,85 EGSM 900,-57,0,54-54
OK

AT+CGDCONT?
AT+CGDCONT?
+CGDCONT: 1,"IP","sl2sfr","0.0.0.0",0,0,0,0
+CGDCONT: 2,"IPV4V6","ims","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,0
+CGDCONT: 3,"IPV4V6","","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,1
+CGDCONT: 4,"IPV4V6","internet","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,0
OK

AT+CDGCONT=1, "IP", "sl2sfr"
AT+CDGCONT=1, "IP", "sl2sfr"
ERROR

AT+NETOPEN
AT+NETOPEN
OK
+NETOPEN: 0

snowgato, 2025-06-28:
AT+CGDCONT?
+CGDCONT: 1,"IP","sl2sfr","0.0.0.0",0,0,0,0
+CGDCONT: 2,"IPV4V6","ims","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,0
+CGDCONT: 3,"IPV4V6","","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,1
+CGDCONT: 4,"IPV4V6","internet","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,0
OK
AT+CGPADDR=1
AT+CGPADDR=1
+CGPADDR: 1,10.10.144.113
OK
AT+UPING="8.8.8.8"
AT+UPING="8.8.8.8"
ERROR

snowgato, 2025-06-28:
AT+HTTPINIT
AT+HTTPINIT
OK
AT+HTTPPARA="URL","http://example.com"
AT+HTTPPARA="URL","http://example.com"
OK
AT+HTTPACTION=0
AT+HTTPACTION=0
OK
+HTTPACTION: 0,200,1256

SFR seems to block ping but HTTP request works !
I  made the mistake to connect with AT+CNMP=13 instead of 38
Thanks  a lot for your help !

lewisxhe, 2025-06-28:
You should check the manual.

closed 2025-06-29 by snowgato