Xinyuan-LilyGO/T-SIM7600X #139: Best Buck Convertor to Power a T-SIm7600

droidblastnz, 2025-07-08:
@lewisxhe what is the best way to power the T-SIm7600G-H? I have need to power 12v IR CCTV LED ring, ESP32Cam, shock sensor, TTGO 0.96 Inch OLED White Color Text Display Module, SR501 PIR and the T-SIm7600G-H. Not using the GPS module.

Using a 12v external [adapter](https://www.aliexpress.com/item/4000760853389.html?spm=a2g0o.productlist.main.9.1f95f158RQISRW&algo_pvid=80e4139e-6611-459c-ab53-ab0204ff7675&algo_exp_id=80e4139e-6611-459c-ab53-ab0204ff7675-8&pdp_ext_f=%7B%22order%22%3A%2225%22%2C%22eval%22%3A%221%22%7D&pdp_npi=4%40dis%21NZD%2116.60%2116.21%21%21%219.78%219.55%21%402101ec1f17519664490858726e582b%2110000007560300620%21sea%21NZ%21121017126%21X&curPageLogUid=GdIKAE8e5KjI&utparam-url=scene%3Asearch%7Cquery_from%3A) and a [Buck Convertor](https://www.aliexpress.com/item/1005006036679284.html?spm=a2g0o.productlist.main.26.67e2630afo7XCU&algo_pvid=93c15584-d0e5-40b4-b584-794938386359&algo_exp_id=93c15584-d0e5-40b4-b584-794938386359-25&pdp_ext_f=%7B%22order%22%3A%2268%22%2C%22eval%22%3A%221%22%7D&pdp_npi=4%40dis%21NZD%215.84%215.67%21%21%213.44%213.34%21%402101c59517519663041011576e0151%2112000035426632599%21sea%21NZ%21121017126%21X&curPageLogUid=LTWl8gipebma&utparam-url=scene%3Asearch%7Cquery_from%3A).

droidblastnz, 2025-07-09:
> The power parameters seem to be fine. But I noticed that you said that the SIM7600 was connected to the computer and ran continuously for a month without any problems.
> 
> The problem only occurred after disconnecting the computer and using an external power adapter, right?
> 
> Where did you connect the power input to the SIM7600?

T-Sim is via a small USB cable from the T-SIM to the Buck Convertor USB C

lewisxhe, 2025-07-09:
It doesn't look like there is any problem.

Are you sure the external step-down module can provide instantaneous 2A current?

Is the USB-C cable capable of passing this current? If the contact points of the USB-C are not in contact, there will be problems.

From the above communication, I guess it is a power supply problem, because you said that there was no problem connecting to the computer for a month, and the problem only appeared after using an external power adapter.

From the picture above, you can see that the power module has a screw installation method. You can connect 5V to the VBUS pin of the T-SIM. The welding method is more reliable. You can try it.

lewisxhe, 2025-07-09:
Well, this cable is of very poor quality. You can solder the power supply according to my method above and connect it from VBUS