Xinyuan-LilyGO/T-SIM7600X #140: Problem with HTTP/HTTPS POST on SIM7600G-H

FlorentThio, 2025-07-29:
Hello,

I am using a SIM7600G-H module (P/N: S2-1097D-Z31E3) in an IoT project and I need to perform secure HTTPS POST requests using AT commands.

Currently, my module runs the firmware LE20B04SIM7600G22.
However, all HTTP/HTTPS AT commands (such as AT+CHTTPSSTART, AT+CHTTPSOPSE, AT+HTTPINIT, AT+HTTPACTION) still return ERROR.

From user and developer feedback, it appears that only firmware versions V4.x.x or newer (typically dated 2023 or later) fully support reliable HTTPS POST operations via AT+CHTTPS* and AT+HTTP* commands.

Could you please provide me with:

The latest firmware version (V4.x.x or newer) compatible with SIM7600G-H (with HTTPS/CHTTPS improvements/bugfixes)

Any required certificates or supporting files

The official update procedure for flashing the new firmware (step-by-step instructions or a guide)

The release notes/changelog mentioning HTTPS/CHTTPS support or improvements

This update is essential for my application, and I want to make sure I follow the correct steps for a safe upgrade.

Thank you very much for your assistance!

> ATI
>
>Manufacturer: SIMCOM INCORPORATED
>Model: SIMCOM_SIM7600G-H
>Revision: SIM7600G_V2.0.2
>IMEI: 862636054511579
>+GCAP: +CGSM
>
>OK
>AT+SIMCOMATI
>
>Manufacturer: SIMCOM INCORPORATED
>Model: SIMCOM_SIM7600G-H
>Revision: LE20B04SIM7600G22
>QCN: 
>IMEI: 862636054511579
>MEID: 
>+GCAP: +CGSM
>DeviceInfo: 173,170

lewisxhe, 2025-07-29:
https://github.com/Xinyuan-LilyGO/T-SIM7600X/issues/135#issuecomment-3017929233

FlorentThio, 2025-08-04:
I recently received the firmware package LE20B05SIM7600G_250425.rar from SIMCOM team as the recommended update for my SIM7600G-H module. However, I noticed that this version differs from the one referenced in LilyGO’s official update procedure for the T-SIM7600G-H board.

To ensure proper compatibility, could you please confirm whether LE20B05SIM7600G_250425 is fully compatible with my hardware, specifically the module with Part Number: S2-1097D-Z31E3, integrated on a LilyGO ESP32-based board?

If not, should I instead follow the firmware version and update method provided by LilyGO?

I appreciate your clarification, as this will help us finalize our integration and ensure HTTPS POST functionality is stable for our project.

lewisxhe, 2025-08-05:
I think the document here already has how to update and update the firmware corresponding to the PN number https://github.com/Xinyuan-LilyGO/T-SIM7600X/blob/master/docs/UpdateGuide/UpdateGuide.md

lewisxhe, 2025-08-06:
Look here
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/HttpsBuiltlnPost

FlorentThio, 2025-08-07:
I need to include a JWT token for authentication, but I'm facing a serious limitation:

🔹 My token is 832 characters long
🔹 The SIM7600G-H seems to have a 512-byte limit for either the HTTP header or POST body

I’ve tried:

Sending the token in the Authorization header → fails due to size
Sending the token in the POST body → also fails if the body exceeds ~500 bytes
Compressing the token with gzip + base64 → ends up even larger (~900 characters)
Has anyone encountered this issue? Is there:

A firmware update that increases the buffer size?
A workaround to send larger payloads (e.g. chunked transfer, proxy relay)?
A way to reduce the token size without compromising security?

lewisxhe, 2025-08-08:
SIM7600G supports 153600 bytes of post data. Have you tried using the post method of http.bin? http.bin will return the post data as is.
[image]

lewisxhe, 2025-08-08:
.
```bash
« AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT

RDY
AT

OK

AT+CPIN?
AT

OK

+CPIN: READY
SIM card online
AT+CNMP=2
AT+CNMP=2

OK
AT+CNMP?
AT+CNMP?

+CNMP: 2

OK
Current network mode : AUTO
Wait for the modem to register with the network.AT+CGREG?
AT+CGREG?

+CGREG: 0,2

OK
AT+CSQ
AT+CSQ

+CSQ: 99,99

OK
[24] Signal Quality:99
AT+CGREG?
AT+CGREG?

+CGREG: 0,2

OK
AT+CSQ
AT+CSQ

+CSQ: 27,99

OK
[25] Signal Quality:27
AT+CGREG?
Registration Status:-1
AT+CGREG?
AT+CGREG?

+CGREG: 0,2

OK
AT+CSQ
AT+CGREG?

+CGREG: 0,2

OK
[29] Signal Quality:99
AT+CGREG?
AT+CGREG?

+CGREG: 0,2

OK
AT+CSQ
AT+CSQ

+CSQ: 27,99

OK
[30] Signal Quality:27
AT+CGREG?
AT+CGREG?

+CGREG: 0,2

OK
AT+CSQ
AT+CSQ

+CSQ: 27,99

OK
[31] Signal Quality:27
AT+CGREG?
AT+CGREG?

+CGREG: 0,1

OK
Online registration successful

Registration Status:1
AT+CPSI?
AT+CPSI?

+CPSI: LTE,Online,460-11,0x775C,117004595,399,EUTRAN-BAND3,1650,5,5,-98,-892,-591,12

OK
Inquiring UE system information:LTE,Online,460-11,0x775C,117004595,399,EUTRAN-BAND3,1650,5,5,-98,-892,-591,12
AT+NETOPEN
AT+NETOPEN

OK

+NETOPEN: 0AT+IPADDR

AT+IPADDR

+IPADDR: 10.4.20.106
Network IP:10.4.20.106

AT+SIMCOMATI

OK

OK

Please make sure you are using the latest released version of the firmware. Find the latest version here: https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/docs/update_fw.md
If you still have problems with the latest firmware, please open an issue. Otherwise, please do not create meaningless issues.
AT+HTTPTERM
AT+SIMCOMATI

Manufacturer: SIMCOM INCORPORATED
Model: SIMCOM_SIM7600G-H
Revision: LE20B04SIM7600G22
QCN: 
IMEI: 860371050882459
MEID: 
+GCAP: +CGSM
DeviceInfo: 173,170

OK
AT+HTTPINIT
AT+HTTPTERM

ERROR
AT+CSSLCFG="sslversion",0,4
AT+HTTPINIT

OK
AT+HTTPPARA="URL","https://httpbin.org/post"
AT+HTTPPARA="URL","https://httpbin.org/post"

OK
AT+HTTPPARA="USERDATA","Accept-Language: zh-CN,zh;q=0.9,en;q=0.8,en-GB;q=0.7,en-US;q=0.6"
AT+HTTPPARA="USERDATA","Accept-Language: zh-CN,zh;q=0.9,en;q=0.8,en-GB;q=0.7,en-US;q=0.6"

OK
AT+HTTPPARA="USERDATA","Accept-Encoding: gzip, deflate, br"
AT+HTTPPARA="USERDATA","Accept-Encoding: gzip, deflate, br"

OK
AT+HTTPPARA="ACCEPT","application/json"
AT+HTTPPARA="ACCEPT","application/json"

OK
AT+HTTPPARA="USERDATA","User-Agent: TinyGSM/LilyGo-A76XX"
AT+HTTPPARA="USERDATA","User-Agent: TinyGSM/LilyGo-A76XX"

OK
AT+HTTPDATA=1226,10000
AT+HTTPDATA=1226,10000

DOWNLOAD
-----BEGIN CERTIFICATE-----
MIIDWjCCAkKgAwIBAgIVAKk3YSBywLpkaZEtlHHVQEEE0axlMA0GCSqGSIb3DQEB
CwUAME0xSzBJBgNVBAsMQkFtYXpvbiBXZWIgU2VydmljZXMgTz1BbWF6b24uY29t
IEluYy4gTD1TZWF0dGxlIFNUPVdhc2hpbmd0b24gQz1VUzAeFw0yMzExMjgwODUw
NDFaFw00OTEyMzEyMzU5NTlaMB4xHDAaBgNVBAMME0FXUyBJb1QgQ2VydGlmaWNh
dGUwggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQDoyGwFHeitOxD123Qj
XDEJ+G3tJHRL+9uWEA55Oqiui4JR5h8lpszUnF4DZ44oRcQAQDh9JOSReunCIEmR
h4PEnYqbaKx4n/bhb9rLybx6n7HlIfOE2ZwwXjNNRiPFqVDr6HGWBozofDcjCRjw
F9gxIZDLZ9bM/bR13wwgb8OPXO38I/eeUwmozMHIc2Y+g6lRgU5WHT1AiYmYBvTQ
+xFTeK6l7HjZdoPQrQDid4SXX+8AggYLTjxxYtzDA/xEdqYI40emEpxbVPa6EBW+
yELaOzP7+S7i5DH3FqDVR24zQllCKj4dDFNM2swbc5BBJDYoYVTp6UTcqa3SEWhP
4aPbAgMBAAGjYDBeMB8GA1UdIwQYMBaAFCjiAGPdhiLiGrKsLq85UMHX6awHMB0G
A1UdDgQWBBR7q08p6D0Oh96G8iC7zna3Dy7ADDAMBgNVHRMBAf8EAjAAMA4GA1Ud
DwEB/wQEAwIHgDANBgkqhkiG9w0BAQsFAAOCAQEAlpCykqgZp90LbrsA+CTzz+Gl
G+ZP/b6A0LOGnyterUj6bXcYx7hGmKyCqHqng/V0hLVGcAG3j1F0PxxmhOIUOJ3R
1lhbSqtvuA2paxaewYfaQSqWVbsEPYhvo0s4Oc3syna7OhdUVmQhFuCaMsOzbLLn
CfV1/zUNMsdHBuqnKYtW3EJcu13Nsat7N4a1a+LjZWMhqjFlsngOW5Vi3ceXL1yP
GzUVD+ZLgcqI6U1oh9u927X17rDuon/B9zkdzHW/GV+pkutEq6hr27jAbl+4TNa9
EI8F6olechhOpFcsgoLyo7bUvKOxINTG/qvKF3wcPcobtciSV/ffujGQreePiw==
-----END CERTIFICATE-----

```