Xinyuan-LilyGO/T-SIM7600X #142: Question about quiescent current SIM7600G-H

acarl0468, 2025-08-04:
Hi,
I have a question regarding the “sleep.ino” example. What current value should I expect with an original SIM7600G-H (T54_V1.3 22-6-27) if I use the program 1:1? I currently have approx. 20mA, which is still much too high for me.
I am currently supplying my board with 3.8V directly from the battery terminals, the SIM card is inserted, and I have only connected the LTE antenna.
Thank you!
Andreas

lewisxhe, 2025-08-05:
The sleep example puts the modem into sleep mode, rather than shutting it down. This completely shuts it down, resulting in a sleep current of 200uA.

https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/fa1a40ce9c2a86c96b5faffa1b9bef4ec2f14877/examples/DeepSleep/DeepSleep.ino#L11

acarl0468, 2025-08-05:
Thanks, that helps me a lot. I simply overlooked the other repository... :-(